cfr34k/t-echo-lora-aprs #1: How to Flash?

DG1FI, 2022-07-09:
I have compiled the Firmware succesful. What is ja good way or Tool to Flash the Firmware? 

73 Mario DG1FI

cfr34k, 2022-07-10:
You should be able to flash it using the pre-installed bootloader of the T-Echo, so there is no need for a special debug interface. I just verified this at the B26 field day with multiple devices. Here is a short summary, please do also read the [_Flashing_ section of README.md](https://github.com/cfr34k/t-echo-lora-aprs#flashing-the-firmware).

You can enter bootloader mode by pressing the reset button twice. Then the red LED should flash rapidly and a storage device should appear when you connect the T-Echo to a PC. Run `make uf2_sd` and copy `_build/nrf52840_xxaa_with_sd.uf2` to the bootloader storage. After that, the device should boot the new firmware.

You can also download a [pre-compiled release](https://github.com/cfr34k/t-echo-lora-aprs/releases). Make sure you are using the `with_sd` version for the first-time installation, otherwise the firmware will not boot.

If there are any issues, please let me know; especially if the firmware is not booting or hangs!

cfr34k, 2022-07-10:
Just one additional note: after flashing you also need to set your call sign via Bluetooth LE before you can enable the tracker and transmit LoRa-APRS packets. Unfortunately this process is currently quite manual and not documented yet, but I will add a description to the Readme during the next days.

73 Thomas DL5TKL

DG1FI, 2022-07-15:
Flash to Device works fine. Thank you...

can you short describe, how can i setup a attribute with [nRF Connect for Mobile]?

DG1FI, 2022-07-15:
Okay i set the callsign characteristic in "UTF-8" format, another text value is not possbile - is that correct? 

The board does not have a gps fix - seems another board layout to your precompiled image?

cfr34k, 2022-07-18:
Yes, "UTF-8" is correct. You can verify that your callsign was set correctly in the „Info“ menu.

Did you manage to get a GPS fix in the meantime? Did it ever work, for example with the default firmware (Meshtastic/SoftRF)?

As one of my boards is also a bit picky about the GNSS signals (does not establish a fix at my QTH), please try the following:

- Leave the „GNSS warmup“ mode on under free sky for at least 30 minutes. Make sure the tracker is off, as the LoRa transmission will probably disrupt the just-established fix again.
- If that does not work, try the same in a different location (some kilometers away).

If it still does not work, I think we have to check the pin assignment/board revision.

DG1FI, 2022-07-30:
Hi! Back again from holidays and now time to test your cool stuff. Tracker works now fine, i must enable tracker and gnss warmup in the menu. Problem/BUG: after pressing reset button, these options are still disabled?! How can i turn the device off? 

73 Mario DG1FI

cfr34k, 2022-07-31:
Hi Mario,

great to hear that the tracker now works for you :)

It should also work without _GNSS warmup_ because the tracker always enables the GNSS module.

Regarding the reset button: yes, it is intentional that everything is disabled after the reset. My expectation of a reset is that the device is in a known, safe state afterwards, which in this case includes that it is not transmitting anything. This is especially important after firmware updates (which the firmware cannot distinguish from a normal reset using the button).

Regarding the power-off: Unfortunately, the hardware does not allow to power off the device completely; there simply is no transistor or other switch that can disconnect the battery. It’s designed by LilyGo as an always-on device. However, I did my best to reduce power consumption as much as possible: if you disable all options (Receiver, Tracker, GNSS warmup) all modules that the hardware allows to be turned off will be turned off. That includes the e-paper display which is only powered during refresh.

Some technical details: there seem to be (at least) two assembly variants: one allows to disable the 3.3V LDO and the nRF52840 (main processor and BLE-SoC) is directly powered from the battery while in the other the nRF52840 is powered from the 3.3V LDO which is always on. In the first variant I could achieve about 0.1 mA standby current consumption and in the other 1 mA.

The only extra thing that could be disabled is the BLE connectivity, but the current consumption from that is minimal (maybe 10 μA of the 100 μA or 1 mA total standby current). So a special „power-off mode“ would be fake and would still drain the battery in some weeks or months.

So unfortunately I have to say: if you want to protect your battery from deep cycling you have to physically disconnect it from the device or recharge it periodically. Harmful deep discharge is luckily prevented by the protection circuit in the battery itself.

73 Thomas

cfr34k, 2022-09-09:
Hi Mario,

do you have any further questions? If not, please close this issue.

73 Thomas

DG1FI, 2022-09-09:
No... I use you new Version, works fine

73

closed 2022-09-09 by DG1FI

uw5elk, 2023-01-13:
T-echo stopped showing up in device manager. Shown only as TECHOBOOT partition when double-clicking the upper side button. Anyone have any ideas?

cfr34k, 2023-01-14:
That’s ok actually if you have installed my firmware, because it does not have any USB functionality implemented and therefore the T-Echo won’t show up when you plug it into a PC.

Double-clicking the upper button launches the bootloader, which is separate from my firmware. The bootloader emulates the TECHOBOOT partition that you have seen.

If you have any further questions, please open a separate issue.