cfr34k/t-echo-lora-aprs #2: Feature question

Gymnae, 2022-08-14:
Hi there,
I was testing out this firmware over the weekend and I appreciate the time you took to lay out the instructions, making it accessible for everyone.

Would it be possible to enhance the status text to include variables like BME280 or battery voltage data? That could be interesting for when the tracker is set to fly on a balloon or just when testing out its features. The T-Echo has some sensor onboard and it would be nice to be able to share some values.

Cheers

cfr34k, 2022-08-15:
Hi, thanks for the suggestions.

Transmitting the battery voltage should be no big deal as that measurement is already available in the firmware. I'll add that in the next version.

I also have a T-Echo with the BME280 so that’s definitely on the roadmap, but I first wanted to have the basic APRS features ready and stable before moving on to the additional hardware.

From my perspective the BME280 needs two major steps:
- implement a driver extracting the measurements from the chip (+ power management). Currently I don't know how tricky the interface of that chip is; I first have to take a look at the datasheet.
- make the data available via APRS. I think there’s a defined format for weather data, but I'd have to read up on that, too.

I'll let you know when there is any progress.

73 de Thomas

Gymnae, 2022-08-17:
Cool, thank you :)
Sorry for the double post.

Side question, would this device also be able to decode radiosondes? If so, it would be very intriguing to have a firmware which could act as both an APRS tracker and weathersonde field tracker.

cfr34k, 2022-08-17:
The [LoRa modem](https://www.semtech.com/products/wireless-rf/lora-core/sx1261) can also receive (G)FSK, so decoding radiosondes might be possible. I don't know how their protocol works, though.

Unfortunately, the time that I can invest into this project is limited, so I can’t support another on-air protocol. APRS is already complex enough and there is still a lot to do (BME280, timekeeping with the RTC, and I think APRS messaging could be fun, too ;-) ).

However, feel free to take this firmware and try implementing it yourself. All the basic structures should be available here.

cfr34k, 2022-08-27:
Hi, I just released version 0.5 with the battery voltage transmission. Please test and let me know if it works for you.

cfr34k, 2022-09-24:
Hi Gunnar,

the just released [Version 0.7](https://github.com/cfr34k/t-echo-lora-aprs/releases/tag/v0.7) now finally has some code to access the BME280 and also to send minimal weather reports via APRS. I'm not 100% sure that this works properly, so please test it and tell me if this is what you had in mind or what can still be improved.

Gymnae, 2022-11-16:
Thank you for the release, I just _now_ had time to test it. The BME280 get recognized, waiting for a fix to send a WX report. I noticed, but already with a different, non-APRS firmware, that the reading of the temperature seems to be off. The sensor might be too close to a piece on the pcb which radiates heat because the temperature is off by at least °7C.
I don't think it's the firmware's fault

cfr34k, 2022-11-19:
Thanks for the feedback. Yes, I noticed that the temperature is too high during tracking as well. Unfortunately, Lilygo placed the BME280 right behind the LoRa module. Also, the heat from the GNSS module (which dissipates about 120 mW) spills over.

Here is a photo of one of my boards as seen from the display side of the device, unfortunately the one without BME280. The BME280 would be placed across the slot below the coin cell battery. The green PCB you can see through the slot is the LoRa module. So the placement is not ideal. However, I think the 120 mW from the GNSS are the major heat source, as the LoRa module consumes less power on average even when tracking.

[image]

Please note that you can trigger a BME280 measurement by refreshing the screen where the sensor values are displayed. Just tap the touch button twice. That works even when the tracker is stopped and GNSS is off.

cfr34k, 2023-09-19:
Hi, I'm just cleaning up the issues in this repo and think this one is basically solved. Therefore, I'm closing it.

Also, in the meantime a lot has changed regarding the weather data handling. Please test and reopen or open a new issue if something does not work as expected.

73 de Thomas DL5TKL

closed 2023-09-19 by cfr34k