cfr34k/t-echo-lora-aprs #5: Looking for help

michaOD, 2022-11-25:
Hello, i am a radio amateur from germany. I am very pleased with this firmware! Great job! Thanks for that.  So far i used a white T-Echo with no problems. Since a few days i am using now a new T-Echo with GRAY case. Everything works except the GPS is always 0/0/0 shown in the display. Unfortunately, i don't know much about programming. Any help or suggestions are aprreciated. Thank you.
[image]
[image]

cfr34k, 2022-12-01:
Hi @michaOD , sorry for the very late answer and thanks for the feedback, I’m glad you like it! So did your GPS work in the end?

michaOD, 2022-12-01:
No, not for now. Do you have an idea/solution?

cfr34k, 2022-12-01:
I’m not sure. I’ve seen this problem before and found a fix for it, see #4 . That worked for all device that I have seen so far.

Does the tracker work despite the status bar showing 0/0/0?

It’s also possible that Lilygo is using a different GNSS module now. All devices I know of are using the L76K module. Is that in yours, too?

michaOD, 2022-12-02:
Hello, thanks for your reply.
Today i tested the tracker again, and it is working fine! I am very pleased about it!
I also opened the T-Echo and it seems to be the L76K Module there. I attached a photo.
I also attached an screenshots of the display.
The number of satellite (0/0/0) and DOP H, V, P, etc. are not displayed.
May this little Informationen helps you to find the cause.
If i can still help you please let me know. Maybe you can find a solution.[image][image]

cfr34k, 2022-12-03:
Ok, thanks for the photos. I'm reopening this issue so others can see that there’s still a problem with the GNSS in case they see this, too.

cfr34k, 2022-12-03:
As you have the same GNSS module as everyone else (that I know of), I’m in the dark why this problem still exists.

Please try the following:

- In the `GNSS utilities` menu, activate `Keep GNSS powered`.
- Then select the `Cold restart` below. It completely resets the GNSS module, so it might take a longer time until it has a fix again.
- Wait until the position is available again (an LED starts blinking on the rear side of the PCB when it’s ready).
- Finally, restart the LoRa-APRS firmware by pressing the reset button. That ensures that the GNSS module is configured correctly (if that works at all).
- Enable the tracker or `Keep GNSS powered` again and check if satellite infos appear.

Hope this helps.

michaOD, 2022-12-05:
I tried as described, but nothing changed. The GPS number is shown as 0/0/0. But fortunately everything else works wonderfully! :)

cfr34k, 2023-03-02:
Seems like there are some more devices that still have this issue. I suspect that there are differently configured GNSS modules around. Unfortunately, I have not seen this issue live since I introduced that the CASIC configuration command is sent as shown here: https://github.com/cfr34k/t-echo-lora-aprs/blob/main/src/gps.c#L147 .

Maybe they switched to PMTK in newer T-Echos?

Here is an experimental firmware build that sends both the PMTK and the CASIC config on startup. Could you try that on your device, please? → [t-echo-lora-aprs-v0.8-1-gd95a532.zip](https://github.com/cfr34k/t-echo-lora-aprs/files/10875005/t-echo-lora-aprs-v0.8-1-gd95a532.zip). Here is the [relevant commit](https://github.com/cfr34k/t-echo-lora-aprs/commit/d95a532c8e6a482d1e1d319f87dcbdbb39e0395f).

cfr34k, 2023-03-05:
Ok, so it’s probably also not MTK-based. Did you reset the T-Echo after the GNSS cold start?

One other thing you could try is to flash different firmware (Meshtastic for example) and check if they can display satellite numbers. If it works I can check what they do differently.

Otherwise I’m a bit out of ideas here right now. I’m not even sure what I could test if I had it on my table. This is more a problem of lacking documentation, I think.

dg3fbl, 2023-03-09:
Yes I did several cold starts, but that didn't help.

The Meshtastic firmware (original wit the device) shows the number of satellites received, but not which type either.

The only thing i can do is to disassemble the device and see which chip it is, or look for another firmware that might give more informations.

Many thanks for your efforts. I would also send you the device if you want to try something.

cfr34k, 2023-03-15:
So, it looks like there are more devices with this problem out there and I now have one from the local community on my table.

I dumped the UART communication of the GNSS module and as suspected, it does not generate all NMEA sentences that it should. Here is the dump starting directly after a reset of the T-Echo:

```
$GPTXT,01,01,02,MA=CASIC*27
$GPTXT,01,01,02,IC=AT6558R-5N-52-1C580901*15
$GPTXT,01,01,02,SW=URANUS5,V5.3.0.0*1D
$GPTXT,01,01,02,TB=2020-03-26,13:25:12*4B
$GPTXT,01,01,02,MO=GR*67
$GNGGA,195233.926,,,,,0,00,25.5,,,,,,*78
$GNRMC,195233.926,V,,,,,,,150323,,,M,V*2E
$GNGGA,195235.000,,,,,0,00,25.5,,,,,,*73
$GNRMC,195235.000,V,,,,,,,150323,,,M,V*25
```

The first line indicates that this is indeed a CASIC device, but is somehow ignores the command to activate additional NMEA sentences. 

On startup, my firmware sends the following sentence that should enable everything necessary according to [the documentation I have](https://espruino.microco.sm/api/v1/files/68b597874e0a617692ebebc6a878032f76b34272.pdf):

```
$PCAS03,1,0,1,1,1,0,0,0,0,0,,,0,0,,,,0*32
```

I don't have an idea yet why this does not work on the newer devices, but I’ll leave this dump here for future reference.

cfr34k, 2023-03-15:
Oh wow, that was easier than expected (for this device at least). I just increased the delay between the GNSS module reset and the configuration from 100 ms to 3 seconds (see https://github.com/cfr34k/t-echo-lora-aprs/commit/3d3b952ae1bf4b63a631f62b170641a90bcd0d18). Now it sends everything as intended :-)

@dg3fbl can you please try the following firmware and check if it works for you, too? 
[t-echo-lora-aprs-v0.8-5-g3d3b952.uf2.zip](https://github.com/cfr34k/t-echo-lora-aprs/files/10984270/t-echo-lora-aprs-v0.8-5-g3d3b952.uf2.zip)

dg3fbl, 2023-03-15:
Great Job! it works now, thank you

cfr34k, 2023-03-16:
Great, thanks for testing. I’ll put this in the next release then :)

cfr34k, 2023-04-07:
I just released [version 0.8.1](https://github.com/cfr34k/t-echo-lora-aprs/releases/tag/v0.8.1) which (among a few other fixes) includes the fix described above.

I close this issue in the hope that the new release solves it for everybody. If there are still problems in this regard, feel free to reopen.

closed 2023-04-07 by cfr34k