cfr34k/t-echo-lora-aprs #6: T-echo stopped showing up in device manager.

uw5elk, 2023-01-14:
T-echo stopped showing up in device manager. Shown only as TECHOBOOT partition when double-clicking the upper side button. How to return to the previous state to flash for example Meshtastic?

uw5elk, 2023-01-14:
Firmware with a .uf2 file causes T-echo to freeze completely.  It is not in the device manager, so I cannot flash it using Platform IO.

cfr34k, 2023-01-17:
The T-Echo can only be flashed through it’s bootloader (the TECHOBOOT partition) or using an SWD debugger (which requires special hardware and opening the T-Echo case). There is no code in my firmware that can trigger the bootloader via USB, so you have to double-click the reset button to make it do anything over USB and show up as TECHOBOOT.

If my firmware just does not boot after flashing, this may have a number of reasons:

1. You used the regular update file without included SoftDevice for the initial flash process. In that case, there is just the wrong version of the BLE stack in the flash, and the firmware cannot boot. Make sure you use `t-echo-lora-aprs-with-sd-v0.8.uf2` (note the `-with-sd` part in the filename!).
2. There is something in the flash that prevents proper startup, for example, leftover configuration data. Such issues should have been solved with version 0.5.
3. You got a new hardware revision that does not work with my firmware.

closed 2023-01-19 by uw5elk