cfr34k/t-echo-lora-aprs #7: How to switch from your firmware to Meshtastic.

uw5elk, 2023-01-17:
Hello.  Please tell me how to switch from your firmware to Meshtastic.  Flashing with a .uf2 file does not complete successfully (still image from your project remains on the display, Meshtastic does not work).

cfr34k, 2023-01-17:
Hi,

you can restore your T-Echo to the original state by flashing the backup of `CURRENT.UF2` you (hopefully) created before flashing my firmware.

If you don't have that backup file, please tell me which Meshtastic .uf2 exactly you are trying to flash, so I can try it on one of my devices. There may be some compatibility issue because my firmware requires and installs a newer SoftDevice (BLE Stack) version than Meshtastic. Therefore a downgrade of the SoftDevice may be necessary before Meshtastic works.

uw5elk, 2023-01-17:
Yes, I don't have a backup of CURRENT.UF2 for Meshtastic.  Your program loads and runs perfectly.  Tried installing any version of Meshtastic and they all end up showing elements of your app on the display.  At the same time, nothing works, except for boot mode when double-clicking the upper side key.  Re-flashing of your program is successful.  You may be able to create a Meshtastic backup file after using your firmware.  Are there any other options possible... Thanks.

cfr34k, 2023-01-18:
That’s unfortunate. I could reproduce the issue and found (so far) that it is only possible to switch back to Meshtastic by flashing a backup file. Flashing an older SoftDevice version (I tried [6.1.1](https://www.nordicsemi.com/Products/Development-software/S140/Download?lang=en#infotabs)) did not help, so probably there is something else missing in the flash that causes Meshtastic to hang.

So, just as a reminder to everyone stumbling across this issue: **Make sure you copy the CURRENT.UF2 from your T-Echo before flashing any custom firmware!**

That being said, I uploaded my own backup file here: [t-echo-backup.zip](https://github.com/cfr34k/t-echo-lora-aprs/files/10450109/t-echo-backup.zip). I could restore one of my devices to Meshtastic using that file, and maybe it works on your T-Echo too, but I can’t guarantee that.

Good luck!

uw5elk, 2023-01-18:
Thank you very very very much!  Your backup saved my T-ECHO.  My fault is that I did not do it before flashing your program.  By the way, it works perfectly (only, as mentioned here, it does not show the number of satellites).

closed 2023-01-19 by uw5elk

dc7os, 2023-03-13:
> That’s unfortunate. I could reproduce the issue and found (so far) that it is only possible to switch back to Meshtastic by flashing a backup file. Flashing an older SoftDevice version (I tried [...]

But i don't get the drive, if i connect the T-ECHO with my laptop to install the current.uf2 :-(

travnewmatic, 2024-01-19:
> [t-echo-backup.zip](https://github.com/cfr34k/t-echo-lora-aprs/files/10450109/t-echo-backup.zip)

really saved my bacon!! thanks so much!