cfr34k/t-echo-lora-aprs #8: How to switch of the device?

dg3fbl, 2023-02-26:
Hello,
how can I turn off the T-Echo? I have tried to do this by pressing the button at the bottom left for a very long time, but it only takes me to the menu.
Is there a screen that is then displayed as "off"?

cfr34k, 2023-02-27:
Hello!

Unfortunately, the T-Echos don’t have the hardware to switch them off completely, so implementing an “off” state would always be fake and therefore I didn’t do that. That being said, I am trying to write this firmware such that it uses the lowest possible amount of power at all times.

That means: if you disable RX and TX (and the „keep GNSS powered“ setting) and don't press any buttons, it is in a very deep sleep state where the SoC consumes microamperes. That includes the BLE advertisement (beacon) transmissions that are still active. Also, the battery voltage is measured once per hour and the display is refreshed so the battery graph stays up to date.

I’ll give some details in case you are interested:

There are multiple hardware or assembly variants for the T-Echo and I have two of them myself:

- The one without BME280 allows to switch off the main 3.3V regulator which powers down everything except the SoC. I could get this device to consume below 100 μA in the deepest sleep state which means the battery will last more than three months.
- The version with BME280 does not allow to switch off the regulator (the Enable pin is wired to always-on), so the LoRa module is always powered and the regulator itself also consumes more power than in the off state. I could not manage to decrease the current below 1 mA, unfortunately, which means that the battery lasts 2 to 4 weeks at best.

There is always a chance that there is still a firmware bug, of course. Driving pins high (or even just leaving a pullup active) while the peripheral device was disconnected from power results in a few hundred microamperes flowing through the peripheral’s ESD protection diodes. The problem is that these issues are really hard to find.

I hope that explains why there is no “off” state. If you have an idea where it would still make sense, let me know.

73 de Thomas DL5TKL

dg3fbl, 2023-02-28:
Hello Thomas,

ok, if the T-Echo doesn't have that, then the question is solved. 

I have the version with the BME sensor, but 2-4 weeks stby is also ok. 
You just don't see that RX and TX are off when you take a quick look at the device. In my opinion, an OFF screen would be nicer here, which can also display the battery voltage.

But that would be a desirable extension if you ever have the time and inclination.

73 Jochen, dg3fbl

cfr34k, 2023-02-28:
Ok, thanks for the suggestion! I’ll think about how to make the device state more prominent when it is idle. Maybe a screen that is shown at the hourly refresh and indicates “TX ON”, „RX ON”, or “Standby” in a larger font? There could be some additional info there, but I still have to think about that. The battery voltage is a good start, though.

closed 2023-03-13 by dg3fbl