cfr34k/t-echo-lora-aprs #9: Fash another bootloader

dc7os, 2023-03-13:
Hi * .*,

i want to change to another firmware and software to test. But how could i set the DFU Mode to get the connected drive on my laptop?

vy 73 de Karsten DC7OS

dg3fbl, 2023-03-13:
Hi Karsten,

wurde hier schon mal beantwortet: https://github.com/cfr34k/t-echo-lora-aprs/issues/7#issuecomment-1387688548

uw5elk, 2023-04-20:
> Hi Karsten,
> 
> to flash another firmware, just double-click the reset button (top left) twice. The red LED should start blinking then and the T-Echo should appear as a storage device on USB where you can drop the UF2 files.

cfr34k, 2023-09-19:
Hi everyone,

I'm closing this issue because there is not really anything I can do here. If you have problems switching to another firmware, please read [the corresponding section in the README file](https://github.com/cfr34k/t-echo-lora-aprs/blob/021894cfab71752ba9d0e85167a1bc5aabcb3373/README.md#how-to-i-switch-to-anotherthe-original-firmware). I have added another link to LilyGo’s documentation regarding re-flashing the bootloader.

73 de DL5TKL

closed 2023-09-19 by cfr34k