cfr34k/t-echo-lora-aprs #10: BME280 data on aprs.fi

matteoramella, 2023-06-14:
'73 Thomas! Thank you for this feature rich firmware. I'm contacting you because I can't display the BME280 data on aprs.fi. I have a T-Echo with the sensor obviously. Displays voltage but not weather data. What am I doing wrong?
Matteo IZ1HMT

matteoramella, 2023-06-14:
Thanks for the quick reply Thomas! Yes, I noticed now that on https://aprs-map.info the weather information is displayed.
I look forward to your next release.
Thank you also for listing the time zone request.
I would also like to translate all the menus into Italian, as I did with other codes found on Github, if you like I can do it for you if you provide me with the part of the code to translate.

cfr34k, 2023-06-17:
Hi again, I have just released [Version 0.9](https://github.com/cfr34k/t-echo-lora-aprs/releases/tag/v0.9.0) which should fix the weather packets and also improves the APRS handling in many other ways. Weather data is now transmitted as _positionless weather reports_, i.e. separately from the position packets. aprs.fi and aprs-map.info correctly show the weather data at the position markers, though. Please let me know if it works for you, too.

I appreciate your offer to translate the menu. Unfortunately, there is no infrastructure for multiple languages in the code so far, so I can't really integrate it right now. I'll think about a good method to implement translation in a general way.

However, if you want to build a dedicated Italian version yourself in the meantime, you can take a look at [menusystem.c](https://github.com/cfr34k/t-echo-lora-aprs/blob/b80b5c2f1561d078aa79fd9a79ed015ad26cc5ff/src/menusystem.c#L490) which contains most of the menu text.

cfr34k, 2023-09-19:
Hi Matteo, could you test version 0.9 and does it work for you regarding weather data? Feedback would be welcome so this issue can either be closed or improved further.

IZ1HMT, 2023-09-20:
As soon as I can, I'm currently using the meshstatic firmware on my T-Echo.

cfr34k, 2023-11-30:
Closing this issue for now. If you find any problems regarding the weather data transmission, feel free to reopen.