cfr34k/t-echo-lora-aprs #11: How to shut down the device?

schulti, 2023-07-29:
Hi! This is an amazing project! Thank you!

Did I miss the function to shut down the device somewhere? I could not find it in the docu. After my tracking I want to shut it down, save battery and recharge. How can I switch it off?

Thank you!

schulti, 2023-07-29:
Hi again! I have just seen that you mention that no "real" shutdown is possible. Maybe a menu item can be added anyhow to shut down all functions (RX, TX?) at once and bring it into the most possible mode for saving power...?

cfr34k, 2023-07-30:
Hi!

> I have just seen that you mention that no "real" shutdown is possible.

Yes, there is no switch (neither physical nor electronic) that allows to switch off the device in a way that it does not draw any current.

> Maybe a menu item can be added anyhow to shut down all functions (RX, TX?) at once

The menu item for a quick standby mode would be possible, but I do not really see the advantage above other options that are already available: You could just hold the touch button and click the lower left button until both TX and RX are off. That is the lowest power mode currently possible and I think doing this is faster than opening the menu and navigating to the „shutdown“ item.

Another option would be to just press the reset once. That is suboptimal, though, because you loose the received packets and the device stays in a not-so-low power state for 3 minutes after the reset.

The only thing that could still be disabled is Bluetooth Low Energy, which sends an advertisement every second even in the lowest power state. Disabling that might save about 5% of the remaining current (i.e. you can leave a fully charged device lying around for 27 instead of 26 weeks before the battery is empty). I'm not sure if that is worth the effort?

diegorispoli, 2023-11-26:
Hi CFR34K, is it still taking a long time to get an update on turning off the device?
Maybe even do the reset button with 3 presses instead of two. Put in the pocket this creates problems.
Thank you very much, development has been stopped since July, I hope we don't have any problems.

diegorispoli, 2023-12-02:
Hi cfr34k, thanks for the reply.
For the reset button it is a good idea to put it in line with the container. To turn off the device, I found the solution by disconnecting the battery with a switch. Everything works fine, when you turn it on you just need to set the tracker to on.
[image]
[image]

diegorispoli, 2023-12-02:
[image]

cfr34k, 2023-12-02:
Yeah, that’s the best solution for a real shutdown 👍

I'm playing around with the software shutdown anyway, because I think it is a good idea to be able to clear the ePaper-Display (all white) before disconnecting the battery for long-time storage. Also I'm interested in how low the current can be in the lowest power mode. We’ll see…

diegorispoli, 2023-12-02:
Do you think it is possible to turn off the two red and blue lights? can it be added to the json configuration file? Do you have any ideas or do you have to go through Arduino?
thanks and good job

cfr34k, 2023-12-26:
Hi, I have just released version 1.0, which includes the software shutdown option.

And sorry, I didn't see your question about the LEDs. However, I'm a bit unsure what exactly you mean, because there are two red and blue LEDs each in my devices. First there is in the firmware-controllable RGB LED which blinks or glows blue occasionally if BLE is active or connected and is red while a packet is transmitted. Then there is the red charging LED and the blue-flashing LED on the PPS output of the GPS. I can surely control the RGB LED, the PPS LED maybe (but very unsure) and the charging LED not at all.

I'm closing this issue now because the original request is resolved, but feel free to comment on the LEDs below and open a new issue if any changes should be discussed.

closed 2023-12-26 by cfr34k