cfr34k/t-echo-lora-aprs #12: Adopt "APL" tocall standard for APRS - LoRa

EA2CW, 2023-08-12:
Now the tocall of the firmware is APZTK1, but almost all LoRa firmwares are adopting the "APL" prefix.
Could be possible to change yours from APZTK1 to, f.i. APLZTK or something like that? 
We at EA2RCF have developed a new map interface of all the the APRS-LoRa resources at:
http://lora.ham-radio-op.net/?center=42.5005,-2.6312&zoom=8
but only the items starting with "APL" will be shown
What do you think?
Thank you very much!
Mikel EA2CW

cfr34k, 2023-08-14:
Hi, yes, I'd love to do that. The APZxxx destinations are for experimental devices and I think my firmware is now major enough to change that. However I'm not sure how I can make sure that there are no conflicts with other firmwares/devices. Do you know if there is some organization that assigns destination calls? Then I could ask there. Thanks! Thomas DL5TKL

EA2CW, 2023-08-14:
Hello Thomas, 
First of all thank you for your work, I love using the Lilygo T-Echo as most simple APRS-LoRa tracker. 

Related to the device-id request, you have an aprs-device id database (and ask for & update protocol) just here in github at:
    https://github.com/aprsorg/aprs-deviceid/tree/main
Which is managing this now. You can find more information about registering new ID's here:
    https://github.com/aprsorg/aprs-deviceid/blob/main/ALLOCATING.md
Hope this is information enough to do it

73, Mikel EA2CW

cfr34k, 2023-08-22:
Hi Mikel, just a short update: I have now [requested a new allocation](https://github.com/aprsorg/aprs-deviceid/issues/94) of the `APLETK` destination call.

EA2CW, 2023-08-22:
Good news! Hope you have success with that!

EA2CW, 2023-09-10:
Finally been asigned APLETK id, congrats!

cfr34k, 2023-09-10:
Yes, very nice :) I have already changed it in the code, so all new builds will use the new tocall.

My plan is to officially release it with version 1.0, together with BLE KISS support (see #13). However, that is quite a big feature and will probably take some time to implement and test.

cfr34k, 2023-12-26:
Ok, I found that the KISS support required too many changes to think about and there were already a lot of smaller changes and fixes queuing up, so I decided to release version 1.0 anyway. So now `APLETK` is the official destination call for this firmware!

Thanks again @EA2CW for your tips!

closed 2023-12-26 by cfr34k