cfr34k/t-echo-lora-aprs #14: TX Power windows entries cutted off

cveith, 2023-09-19:
Hi there,

i'm using Version **0.9** and everything is working fine with it. By the way i'm using the SoftRF version and setting up that version was the same as described on that repo. I think the hints regarding SoftRF can be removed from the readme.md.

One thing i noticed is on the TX Power settings screen. The list is not shown correctly. The "-9 dBm" entry is only visible half.
[image]

best 73s,
Christian, DL5CV

cfr34k, 2023-09-19:
Hi Christian,

thanks for your report.

> The "-9 dBm" entry is only visible half.

Oh, i see. That must have happened a long time ago when the title bar became larger. Maybe I could just remove the Cancel-entry from this screen and pre-select the current power. What do you think?

> By the way is there any way to "protect" the reset button via software? A long keypress to reset the device would be neat.

Unfortunately, the reset button is wired to the default hardware reset of the nRF52840 chip. I'm not sure if the firmware can disable the reset functionality at this pin. Even if it can, doing so is probably a bad idea because if the firmware hangs the only way to reboot it is to disconnect the battery. Also entering the bootloader (double-reset) would be quite hard then.

As a hardware solution you could just file down the button such that it is the same level as the case. Then it should be harder to press it accidentally.

73
Thomas, DL5TKL

cveith, 2023-09-19:
Hi Thomas,

> Hi Christian,
> 
> thanks for your report.
> 
> > The "-9 dBm" entry is only visible half.
> 
> Oh, i see. That must have happened a long time ago when the title bar became larger. Maybe I could just remove the Cancel-entry from this screen and pre-select the current power. What do you think?

I think that would be a viable solution as it's just a confirmation of the current value.

> Unfortunately, the reset button is wired to the default hardware reset of the nRF52840 chip. I'm not sure if the firmware can disable the reset functionality at this pin. Even if it can, doing so is [...]

Yeah filing it down would be the best solution for the moment. I think I'll send an email to lilygo to change the button or case to a version that can't be pressed that easily.

Best regards

Christian, DL5CV

radio-Joe, 2023-09-20:
Hi Thomas, 

an easy solution is to omit the power step +14dBm. No need of such little steps. 

best regards
Josef, dj3gi

cfr34k, 2023-09-20:
Yes, removing one power option would also be possible, of course, but which one? The first four entries (22, 20, 17 and 14 dBm) are from the SX1262 datasheet and are fully specified including the current consumption as optimal settings for the PA. Therefore I would like to keep those. For example, the step between 22 and 20 dBm may seem small, but it results in a reduction of DC current from 107 to 65 mA!

I still prefer to remove the „Cancel“ entry.

cfr34k, 2023-09-24:
I now removed the cancel entry (commit 8865d0530f8b67e3be39b0a764d7692c09553538). As this is just a visual change, I'll release it with version 1.0 once that is ready.

You can try it in the meantime, though: [t-echo-lora-aprs-v0.9.0-4-g8865d05.uf2.gz](https://github.com/cfr34k/t-echo-lora-aprs/files/12708862/t-echo-lora-aprs-v0.9.0-4-g8865d05.uf2.gz)

cveith, 2023-09-25:
Hi Thomas,

i've tested the prerelease so far. It works nice :)

But i haven't tested the power levels and the resulting output power, but i guess it should work properly as nothing in that has changed.

73 de Chris, DL5CV

cveith, 2023-09-25:
Hi Thomas,

sorry to have another bug ;)

You can select any entry properly, go back into the power menu and you'll find that setting activated.

This works for everything except for the +22dBm entry. When you select that one after you've had e.g. -9dBm, the value won't be persisted.

The active usable range is currently +20dBm till -9dBm.

cfr34k, 2023-09-25:
Thanks for testing and reporting back so quickly. I forgot to remove the special handling of the first menu entry as "Cancel". As +22 dBm now is the first menu entry, it still acted like the old "cancel" entry and was never applied.

Fixed in 21fd92a. Here is an updated build: [t-echo-lora-aprs-v0.9.0-5-g21fd92a.uf2.gz](https://github.com/cfr34k/t-echo-lora-aprs/files/12717422/t-echo-lora-aprs-v0.9.0-5-g21fd92a.uf2.gz)

cveith, 2023-09-25:
That did the trick :)

Tnx for that quick fix Thomas.

closed 2023-09-26 by cveith