cfr34k/t-echo-lora-aprs #15: Incorrect rx packet age

cveith, 2023-10-01:
Hi Thomas,

i've found that the rx age is calculated incorrectly, when the packet has been received without a valid time set on the tracker (no gps fix after reset).

[image]

In that case the packet was sent with the tracker having not had a gps fix. The time of the tracker was therefore 1970-01-01 00:02:00.

When the gps was fixed, time on the tracker changed to 2023-10-01 xx:xx:xx.

During display, the stored unix epoche was calculated with the current time which lend to a packet age of over 19k days.

I've forked you're repo and did a fix for that issue in branch fix/invalid_rx_age.
https://github.com/cveith/t-echo-lora-aprs/tree/fix/invalid_rx_age

I could prepare a pull-request for that fix, if you like to.

Best regards

Chris, DL5CV

cfr34k, 2023-10-03:
Hi Chris,

sorry for the late reply. Your changes look good to me and I think this is a nice usability improvement. I'd appreciate a pull request.

Thanks!

Thomas, DL5TKL

cveith, 2023-10-03:
Hi Thomas,
no worries. I appreciate the time spent on private projects.

I've created the pull request and linked it to this issue.

best regards

Chris, DL5CV

PR #16 (2023-10-03): Added rx_time_valid variable
Added rx_time_valid information to rx'ed aprs packets. Updating rx_timestamp once time is valid

fixes #15
closed 2023-10-03 by cfr34k