cfr34k/t-echo-lora-aprs #17: No data via LoRa digipeater

macucuma, 2023-11-03:
The packet sent whith t-echo aren't repeated over rf from digirepeaters because WIDE1-1 is not sent.
RF digipeaters are widely used where there is no possibility of internet but t-echo is ignored

cfr34k, 2023-11-04:
Hi,

yes, WIDE1-1 is no longer sent by this firmware since version 0.9. Instead, Destination Call Digipeating is used, where the number of digipeater hops is sent in the destination call SSID of the APRS packets. You can read about this method in the APRS specification version 1.0.1 on pages 15 and 16. In version 0.9 of this firmware, the packets are addressed to APZTK1-1, where the appended -1 is equal to WIDE1-1. The reason to do this is the reduction of LoRa airtime, which can be a limited resource in crowded areas.

Also, please note that most LoRa-APRS IGates don't digipeat on air even if requested with WIDE1-1. Therefore, sending the request is mostly a waste of airtime. Destination Call Digipeating is a reasonable compromise between the full WIDE1-1 request and no digipeating at all.

macucuma, 2023-11-04:
Many thanks for the explanation 
I close issue

closed 2023-11-04 by macucuma