cfr34k/t-echo-lora-aprs #18: Request: User selectable frequency

AtomicZombie, 2023-11-26:
Hello,

First just wanted to say that I really like this firmware!

In the United Kingdom our spectrum regulator has decided to use 439.9125MHz for LoRa APRS instead of the normal 433.775MHz, more information on this can be found here - [https://reflector.sota.org.uk/t/lora-aprs-tracker/31589/8](https://reflector.sota.org.uk/t/lora-aprs-tracker/31589/8
)

Would it be possible to have the frequency used in this software to be user selectable in some way? Perhaps via Bluetooth?

I guess it would be good to be able to select the SF / BW / CR as well but I'm not sure how many other countries are using 'unique' settings. Us in the UK might be the exception for now.

Right now I've managed to build a version with this different frequency, however this might not work for everyone who might want to use a T-Echo.

Thanks for all your efforts in this project, it really is appreciated. 
Dayle 2E0OUE

cfr34k, 2023-11-30:
Hello Dayle,

great to hear that my firmware has found its way to the UK :) Interesting that you got a different frequency assigned.

Maybe this is a good opportunity to introduce a kind of generic Bluetooth LE settings interface. My current idea is to add two characteristics, one to write or select a setting and the other to read the current value. The first byte identifies the setting that’s accessed. I think that would also be a good way to make all the other options from the menu available via BLE.

I'll see what I can do.

Thomas DL5TKL

cfr34k, 2023-12-16:
Hi Dayle,

the `main` branch now (after bfce77dbdb16e357c955c93e6430eb5f63bc23f2) contains the basic changes needed to configure the LoRa RF frequency over BLE. Documentation is not updated yet, but basically you have to do the following to set the frequency:

1. Express your desired frequency in Hz: 439912500 Hz
2. Encode it into a little-endian integer: `439912500` decimal = `0x1a388834` hex (big endian) → `0x34 0x88 0x38 0x1a` (little endian)
3. Prefix that value with `0x07` (ID of the RF frequency setting) and write everything into the new "Settings write" characteristic (UUID `00000110-b493-bb5d-2a6a-4682945c9e00`): `0x07 0x34 0x88 0x38 0x1a` 
4. The new frequency is applied when the LoRa module is next turned on (you may have to restart the RX).

You can also use the `techo_client.py` script, which already supports the new settings interface.

I would really appreciate if you could test this on your device.

AtomicZombie, 2023-12-17:
Hello Thomas

I've now built and tested the latest version, and I can confirm it works.

After building I used the 'NRF Connect' iOS app to change the required setting. The frequency shows correctly on the device, and I can both send and receive APRS packets.

Let me know if you need any more information / testing.

I really appreciate your work on this project!

Thanks,
Dayle M0OUE

cfr34k, 2023-12-17:
Great, thanks for the quick response!

Then I'll release it in v1.0 as soon as the other issues are resolved (probably in the next two weeks).

cfr34k, 2023-12-26:
I've just released version 1.0 including these changes, so I'm closing this for now.

If any further changes are needed for the UK or any other country, feel free to reopen or create a new issue!

closed 2023-12-26 by cfr34k

M7DSQ, 2025-08-10:
Hi, I am using in the uk also and I don't have much knowledge of how to program .. is there any chance you can compile the file needed for the uk so it's set at our frequency As I'm really struggling and have been all week trying to change it ... I will of course make a donation .. my email is davechalton34@gmail.com 

Kind regards 

Dave 
M7DSQ 

Ps I have read and re read everything here and just can't get my head around it or get it to change frequency.. 

Thanks again 
Dave