cfr34k/t-echo-lora-aprs #19: Feature Request

dc7os, 2023-11-29:
Is it possible to save the settings "Receiver on/off" and "Tracker on/off" too???

cfr34k, 2023-11-30:
Hi!

Not yet. I also think it would be better to be able to configure a default state (for example: TX on, RX off) that is activated after reset than to just restore the previous state. Would that be ok for you?

73 de Thomas DL5TKL

cfr34k, 2023-12-26:
Hi,

I just released v1.0 with the new `APRS Config`→`Advanced`→`Startup state` menu item that allows to configure the receiver and tracker state on device startup. Please try it and let me know if it works for you.

For now I'm assuming this issue is resolved. Feel free to reopen if you think otherwise.

73 de Thomas DL5TKL

closed 2023-12-26 by cfr34k