cfr34k/t-echo-lora-aprs #21: Missing "wide1-1" for digipeating

IZ1HMT, 2024-05-05:
Please can you implement the "wide1-1" string in the firmware?

cfr34k, 2024-05-07:
WIDE1-1 was included some time ago, but was removed in favor of _destination call digipeating_. See https://github.com/cfr34k/t-echo-lora-aprs/commit/b39ec4d9f6469aacebe47e4f6c6bcb2ec857ff16

As digipeating is usually not done for LoRa-APRS and airtime is a precious resource, this is a reasonable compromise. If you really need WIDE1-1, you can just revert the above commit.

IZ1HMT, 2024-05-09:
I now notice that neither the firmware of IW1CGW https://github.com/iw1cgw/LoRa_APRS_Meteo_dds-cgw nor that of Ricardo CA2RXU https://github.com/richonguzman/LoRa_APRS_iGate support this mode of DIGI. I think in the next firmware it would be better if I go back to the old wide1-1 to maintain better compatibility. With your setting which DIGI firmware do you use to make the beacon repeat correctly?

cfr34k, 2024-05-09:
I have never actually seen a digi/iGate that does digipeating on LoRa APRS, even when WIDE1-1 was still included, so I assumed it was not really relevant. That’s also what I hear from other users with different devices. At least here in Germany digipeating does not seem to be very popular. Maybe that’s different in other countries?

However, I still don't really like switching back to WIDE1-1 by default, as it increases the airtime by about 10% for every packet (which is a lot considering the channel is accessed randomly by many users). However, I can add an option that allows to select the addressing mode from:

- no digipeating (`>APLETK`)
- destination call digipeating (`>APLETK-1`)
- WIDE1-1 (`>APLETK,WIDE1-1`)

Not sure yet if no digipeating or destination call digipeating will be the new default.

> I will do my best to ensure that the developers of these firmware introduce support.

That would be really nice!

IZ1HMT, 2024-05-09:
Yes, in Italy "pure" Dgipeaters are popular, especially in the mountains and where there is no internet connection. So that the beacons will be picked up by the first available iGate. It would be great to have a menu option like you described. At the moment I am forced to use the old 0.8.1 due to my difficulties in compiling a modified firmware.

cfr34k, 2024-05-19:
Hi, I have now added a configuration option as described in my last post. You can find it in the APRS Advanced configuration menu.

I could not test it yet on the device, as I don't have it nearby, but I'm quite confident that it works ([the changes](https://github.com/cfr34k/t-echo-lora-aprs/commit/9e6251fa91fb95a2fb700981f8e9867630ba19e8) were not too large). Here is a build, feel free to test:

[t-echo-lora-aprs-with-sd-v1.0-2-gcc438bf.uf2.gz](https://github.com/cfr34k/t-echo-lora-aprs/files/15370230/t-echo-lora-aprs-with-sd-v1.0-2-gcc438bf.uf2.gz)

IZ1HMT, 2024-05-19:
Thanks so much! Now I'm on holiday with my t-echo but I don't have a computer available. When I return in a few days, I will try the new firmware and update you.

DayleDrinkwater, 2024-05-19:
Hi,

I've been following this in the background. I've just tried the new version and it seems to be working. My digipeater is digipeating the packets correctly and they are appearing on the  T-Echo.

Thanks,
Dayle M0OUE

IZ1HMT, 2024-05-20:
I found a computer and installed the file. It works perfectly.

closed 2024-05-20 by IZ1HMT

cfr34k, 2024-05-31:
Sorry for the late reply. Thanks a lot for testing. I have released this change in [Version 1.1](https://github.com/cfr34k/t-echo-lora-aprs/releases/tag/v1.1).