cfr34k/t-echo-lora-aprs #22: techo_client.py : TypeError: 'NoneType' object is not subscriptable

stefaon, 2024-07-06:
Hello,
It's not possible to set callsign in the t-echo with the script, because there is an error.
I use python 3.12 on macosx.

the error is :
Python 3.12.4 (v3.12.4:8e8a4baf65, Jun  6 2024, 17:33:18) [Clang 13.0.0 (clang-1300.0.29.30)] on darwin
Type "help", "copyright", "credits" or "license()" for more information.

= RESTART: /Users/stephan/Downloads/t-echo-lora-aprs-main/tools/ble_client/techo_client.py
Scanning for 5 seconds...
Traceback (most recent call last):
  File "/Users/stephan/Downloads/t-echo-lora-aprs-main/tools/ble_client/techo_client.py", line 164, in <module>
    asyncio.run(main())
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
  File "/Users/stephan/Downloads/t-echo-lora-aprs-main/tools/ble_client/techo_client.py", line 65, in main
    if d.name[:6] == "T-Echo":
TypeError: 'NoneType' object is not subscriptable

73's de F1UBL.

stefaon, 2024-07-08:
Hello,
Thanks for the reply.
It's better, I had a new error, but I see my T-ECHO.

Python 3.12.4 (v3.12.4:8e8a4baf65, Jun  6 2024, 17:33:18) [Clang 13.0.0 (clang-1300.0.29.30)] on darwin
Type "help", "copyright", "credits" or "license()" for more information.

= RESTART: /Users/stephan/Documents/Radio/TTY APRS/TTY T-ECHO/t-echo-lora-aprs-main/tools/ble_client/techo_client.py
Scanning for 5 seconds...
Found the following T-Echos:
0 = [1A7B6394-3162-4B0F-B968-7FBA09AD6308] T-Echo F1UBL-7
Type the entry number to connect: 0
You selected #0: 1A7B6394-3162-4B0F-B968-7FBA09AD6308: T-Echo F1UBL-7
Connecting...
Traceback (most recent call last):
  File "/Users/stephan/Documents/Radio/TTY APRS/TTY T-ECHO/t-echo-lora-aprs-main/tools/ble_client/techo_client.py", line 164, in <module>
    asyncio.run(main())
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
  File "/Users/stephan/Documents/Radio/TTY APRS/TTY T-ECHO/t-echo-lora-aprs-main/tools/ble_client/techo_client.py", line 80, in main
    is_paired = techo.details['props']['Paired']
TypeError: tuple indices must be integers or slices, not str

By the way, I find how to set my call with a phone and the application you mentionning into the documentation.
So it's work for me actually, it's nice tracker, thanks for that.
73's de Stephan

cfr34k, 2024-07-09:
I just verified the script works on my Linux system, so it must be some difference in your setup. From the paths in your backtrace I conclude that you are running running some other operating system (Windows?). As far as I know, nobody has ever tested the script on other systems than Linux, so we have to work through the details here together (I don't have a Windows installation to test with).

As you could find the T-Echo with the script, that’s a good start already, because the `bleak` Python library works. To debug the above issue, could you please add the following line and post the output here?

```patch
diff --git a/tools/ble_client/techo_client.py b/tools/ble_client/techo_client.py
index 5c9ef24..b289b6c 100755
--- a/tools/ble_client/techo_client.py
+++ b/tools/ble_client/techo_client.py
@@ -77,6 +77,7 @@ async def main():
     print("Connecting...")

     techo = techos[idx]
+    print(techo.details)
     is_paired = techo.details['props']['Paired']

     client = BleakClient(techo)
```

You can also try to replace the line `is_paired = techo.details…` with just `is_paired = False` (and answer the question about pairing with `n`). You should then be able to show the current configuration at least.

> So it's work for me actually, it's nice tracker, thanks for that.

You’re welcome :)

stefaon, 2024-07-10:
Hello,
This is the debug :

Python 3.12.4 (v3.12.4:8e8a4baf65, Jun  6 2024, 17:33:18) [Clang 13.0.0 (clang-1300.0.29.30)] on darwin
Type "help", "copyright", "credits" or "license()" for more information.

= RESTART: /Users/stephan/Documents/Radio/TTY APRS/TTY T-ECHO/t-echo-lora-aprs-main/tools/ble_client/techo_client.py
Scanning for 5 seconds...
Found the following T-Echos:
0 = [4F98E381-FE3B-4307-BCAE-2F39C5CDCF6B] T-Echo F1UBL-7
Type the entry number to connect: 0
You selected #0: 4F98E381-FE3B-4307-BCAE-2F39C5CDCF6B: T-Echo F1UBL-7
Connecting...
(<CBPeripheral: 0x7fdd8e319ce0, identifier = 4F98E381-FE3B-4307-BCAE-2F39C5CDCF6B, name = T-Echo F1UBL-7, state = disconnected>, <CentralManagerDelegate: 0x7fdd8e30f230>)
Traceback (most recent call last):
  File "/Users/stephan/Documents/Radio/TTY APRS/TTY T-ECHO/t-echo-lora-aprs-main/tools/ble_client/techo_client.py", line 165, in <module>
    asyncio.run(main())
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
  File "/Users/stephan/Documents/Radio/TTY APRS/TTY T-ECHO/t-echo-lora-aprs-main/tools/ble_client/techo_client.py", line 81, in main
    is_paired = techo.details['props']['Paired']
TypeError: tuple indices must be integers or slices, not str

I dont know if it's important : I use it on my Macbook Pro MacOSX 10.15.7

73's

cfr34k, 2024-07-20:
Sorry for the late reply.

Ok, so `techo.details` looks completely different compared to my Linux OS. Then I don't really know how to implement this pairing check such that it works for all platforms.

You can try to just set `is_paired = True` instead of the problematic line and it might work if you do the pairing in the OS. Even without pairing, you should then be able to read the settings from the T-Echo.

stefaon, 2024-07-23:
Hello,
Thanks,
I try : it work fine !
I was able to set the infos via the python script by this way.

73’s de Stephan.

Cordialement,
Stéphan ROBERT