cfr34k/t-echo-lora-aprs #24: Software does not find the GPS

2e0wkw, 2024-08-29:
When running this software (V 1.1) it cannot get a fix on the GPS and so will not track. I know the GPS to be fine on this unit.
I have seen that there have been a great many issues with T-Echo and GPS with other LoRa software (Meshtastic) are we suffering the same issues here.

I think this is a great project and I would love to be able to use it.

Eric
M0REQ

cfr34k, 2024-09-01:
Hi, that seems to be either a hardware issue, GPS module misconfiguration or you have a different module in your device.

Did you already do a cold restart of the GPS module (see the [documentation](https://github.com/cfr34k/t-echo-lora-aprs/blob/main/doc/manual.adoc#gnss-utilities)? Afterwards, please leave the GPS or tracking active for at least 30 minutes under open sky.

If it does not achieve a fix then, try again a few hundred meters away. One of my devices shows the weird effect that it does never find a position nearby my flat, but works instantly when I go for short walk. The other device receives >10 satellites inside the flat, so that is not a interference problem, but a bug in the GPS module.

If that does not help, please check that your device has indeed an L76K GPS module inside.

Thomas
DL5TKL

2e0wkw, 2024-09-03:
Hi, yes it might be a H/Q issue. I have ordered another to see.
Cold Restart - check
30 mins in open sky - check
L76K Modile - check

I have not tried moving away from the house yet.
I have also opened up the unit and check that the GPS Antenna is connected properly.
Given that the T-Echo is known to have GPS issue maybe we need more debug/test tools here.

Eric M0REQ

cfr34k, 2024-09-03:
Ok, so there really is a hardware issue or serious misconfiguration. Did you check if the module is supplied correctly?

If you want to debug the communication on the low level, I think it would be best to get a dump of the UART signals. I see two options:

1. Read the communication directly from the data lines using an USB-UART-Adapter or Logic Analyzer. I think there are diodes in the path where the probes can be attached.
2. Connect a debugger to the SWD interface and read the logs via JLink RTT. The relevant log entries for the NMEA messages are currently commented out in the code, but I can provide a special firmware if you want to take this path.