cfr34k/t-echo-lora-aprs #25: Problems flashing the firmware.

IV3JYV, 2024-10-26:
Hi,

I tried to flash the firmware copying the UF2 file in the techoboot folder but when it is copying the file at a certain percentage the device is disconnected and nothing is happening. When double clicked the reset button the mass storage device is reappearing but always with current.udf2, therefore the new firmnware is not installed. 
Any idea on how to flash it in an alternative way? I tried all steps many times without success.

DayleDrinkwater, 2024-10-26:
Hi, have you tried with a different USB cable? Also even when successful it will be named current.uf2 anyway

M0OUE

cfr34k, 2024-10-27:
M0OUE is right, you will always only see current.uf2.

Additionally, are you using the `with-sd` version if you are installing my firmware for the first time? This one: https://github.com/cfr34k/t-echo-lora-aprs/releases/download/v1.1/t-echo-lora-aprs-with-sd-v1.1.uf2 . It includes the bluetooth stack which has to be installed for my firmware to work.

IV3JYV, 2024-10-27:
OK with the file at the link the system is rebooting correctly. Now hopefully I can set the device in the right way.

closed 2024-10-27 by IV3JYV