cfr34k/t-echo-lora-aprs #28: Available for confiog: SF and BW

jampac, 2025-02-27:
Thanks for this excellent piece of firmware for the T-Echo!. I am opening this issue as I can see a need to have SF and BW available for config. Can this be implemented either using the techo_client.py or via the UI / config menu. I can see the BW and SF being static defined in lora.c. in SX1262_LORA_SF_BYTE_IDX and SX1262_LORA_BW_BYTE_IDX. Background: a number of hams here are using more suitable SF10 and BW125 away from LIPD (Low interference potential devices call license) interferences.

Really appreciate everyone contributing! Thanks! Morten 73 VK1JAM

cfr34k, 2025-02-27:
Hi! Yes, I think I can add those to the BLE config interface, so you can set it up via `techo_client.py`. As these settings probably do not need to be changed often, it should be ok to have them only there and not in the menu (which is quite full already).

It will take some time to implement this, though, because I didn't work on this project for quite a while.

cfr34k, 2025-03-02:
Hi again! I found some time this weekend to implement this and already have a test version for you. The modulation parameters can now be set through the advanced settings interface (similar to the RF frequency). Details and an example can be found in the [documentation](https://github.com/cfr34k/t-echo-lora-aprs/blob/4103c71dd99ce3d6c0bde1a7fd75d879b8237994/doc/manual.adoc#lora_mod_config).

Here is the update file: [t-echo-lora-aprs-v1.1-7-g1217680.uf2.gz](https://github.com/user-attachments/files/19044941/t-echo-lora-aprs-v1.1-7-g1217680.uf2.gz)

I have also updated `techo_client.py` to include the new parameters. The new version is already [in the main branch](https://github.com/cfr34k/t-echo-lora-aprs/tree/56ebeb63809f74d042f873c81138c1b37920d5f6/tools/ble_client).

Please let me know if this version works with the different SF that you mentioned.

jampac, 2025-03-10:
Thanks a lot for implementing these changes. I have configured using tech.client.py the changes to the Lora modulation and are going to test it - will take a week or some. Thanks!