cfr34k/t-echo-lora-aprs #29: configuring the smart beacon

DO7FIR, 2025-07-06:
thanks for that nice firmware. it works really good. only thing i miss is a point where i can set the smart beacon parameters like slow and fast speed and tx interval. is there a point in the code anywhere? thanks for your reply in any way.

cfr34k, 2025-07-14:
Hi, and sorry for the late reply.

There is currently no user-interface for configuring the beacon parameters. If you can and want to compile the code yourself, you can adjust the parameters in [src/tracker.c](https://github.com/cfr34k/t-echo-lora-aprs/blob/main/src/tracker.c#L39).

DO7FIR, 2025-07-14:
> cfr34k

oh perfect. do you have an explanation for the parameters and whitch values are accepted?

cfr34k, 2025-07-16:
I have just added some comments to [the code](https://github.com/cfr34k/t-echo-lora-aprs/blob/1d9406436c6d03eed58e7688ca9ca2dca6622c49/src/tracker.c#L39). I hope they explain well enough how the parameters work.

If you change them, please do not reduce `MIN_TX_INTERVAL_MS` further, as that may cause congestion of the LoRa APRS channel due to very frequent transmissions.